fdebrabander/Arduino-LiquidCrystal-I2C-library #2: Does not print more than one character.

Avamander, 2015-10-26:
This library only prints the first letter of the string that had been passed to the .write(string) function.

AnDann, 2015-11-07:
Have the same problem here.

Somnitec, 2015-11-26:
Same issue on Arduino 1.6.6. None of the lcd.* functions seem to make a difference.

pprovaz, 2015-12-27:
Hello guys
in LiquidCrystal_I2C.cpp
just add a line "return 1;" after 203. The procedure will look this way:

inline size_t LiquidCrystal_I2C::write(uint8_t value) {
	send(value, Rs);
        return 1;
}

Compile, and it will work fine!
The new Arduino IDE checks the return value of this proc and it has to be 1 to work fine.

Cysign, 2015-12-29:
Hmmm.....same here. I already checked line 203 and (on my fresh re-downloaded LC-i2c-lib) it already is return = 1.

Any other suggestions?

pprovaz, 2015-12-29:
Be sure to remove all other versions from your folders. New Arduino IDE may not use the version in the libraries folder, it uses fist version of the same named file it finds, for example in your sketch folder. It may give you a note in the log, what version does it use.

If this does not help, make this test: Download the file from here or copy all the lines from here and paste them over your file. Do the same for LiquidCrystal_I2C.h. Insert the line 203 as described above. I did this yesterday and it works.

Cysign, 2015-12-29:
Well, can you tell me where the logfile is located or what it's name is, please? (Windows system)

I figured out I have had left a copy of the old files still in the libraries folder (one folder called LiquidCrystal-I2C and another one LiquidCrystal - noe deleted the latter one). Then I ensured that the other one has the changes on line 203 (return = 1).
To make sure it is identical with the library I copied the content for both, the cpp and h files.
Then I restarted the IDE but I still do only get the first letter of my string on the display.

pprovaz, 2015-12-29:
If you used the libraries auto update (built in the new Arduino IDE), one library copy is also in your Documents\Arduino folder.

If there are more versions, it will notify you in the log, located under the code window in Aduino IDE. It is the black area near the bottom of the screen in Arduino IDE.

Cysign, 2015-12-29:
That's it!
Thanks!

A copy has been in C:\Users\<username>\Documents\Arduino\libraries

Can I disable using this folder?
I usually prefer having all neccessary files in the program folder....it's easier to backup if all belongs to one place!

pprovaz, 2015-12-29:
Perfect!

Not sure about disabling the folder. This path is defined in File -> Preferences as a sketchbook location. You may try to leave it blank, not sure what happens.

Cysign, 2015-12-30:
If you leave it blank or fill in an incorrect path, it resets to deafult.
I just filled in a subfolder in ma IDE and now I'm just moving all new downloaded updates to my library-folder and overwrite the old files.
That's more comfortable than always searching the default folder. But I'd prefer them to ask to overwrite the old libs and create a backup of the old files when there's an update available...
Still much potential for improvements on the ArduinoIDE.

Thanks for your help :)

pprovaz, 2015-12-30:
Ok, then I would point it to the library folder of your Arduino installation, or your usual sketch folder.

PramodKarunarathne, 2017-01-07:
Guys...change the clock to Internal 16MHz ;)

Avamander, 2017-01-07:
To make a library work?

dous56, 2017-11-22:
Thank you very much 

#  `pprovaz `#

Your suggestion works very well
Georg

Biohidraulica, 2018-07-03:
Thank you !!

Also, your suggestion works very well!

Jose