fdebrabander/Arduino-LiquidCrystal-I2C-library #3: This library should be renamed

Avamander, 2015-10-26:
I think this library needs a different name as it is not compatible with libraries with the same name.

ThorinDev, 2016-03-17:
I agree, Arduino keeps on wanting me to update to a different library, therefor breaking all my code

WerkAG, 2016-03-17:
I agree too. I'm now using this library, but I have code which uses the other older library with the same name, which causes some problems. Renaming this one will be very nice.

per1234, 2016-04-11:
The issue of the spurious update notifications could be fixed by adding a library.properties file with a unique `name` value(different than any of the other libraries in Library Manager). More information: https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification

ThorinDev, 2016-04-14:
Have created a repo. We should move the project over there because this is abandoned.
https://github.com/ThorinDev/LiquidCrystal-I2C

per1234, 2016-04-14:
>this is abandoned.

What makes you think that? The last commit was December 27th.
>Have created a repo.

Why not fork this one?

tablatronix, 2016-04-15:
Heh last commit just happened, somis this no longer tracked in arduino library manager? I get crc errors or not found when installing.

per1234, 2016-04-15:
>Heh last commit just happened, so is this no longer tracked in arduino library manager? I get crc errors or not found when installing.

@tablatronix I confirm the issue but the LiquidCrystal I2C library in Arduino Library Manager is a different repository. Please open an issue on that repository at: https://github.com/marcoschwartz/LiquidCrystal_I2C/issues/new

tablatronix, 2016-04-15:
So is it a unique name conflict , cause they both said  fdebrabander which i thought was odd. Ill look again tomorrow. Thx

per1234, 2016-04-15:
>So is it a unique name conflict

The problem described in this issue is that when you follow the installation instructions in this repository's Readme since there is no library.properties file the library gets the name LiquidCrystal_I2C from the folder name. Some of the releases in https://github.com/marcoschwartz/LiquidCrystal_I2C, which has been added to the Arduino Library Manager, also have that same `name` value in library.properties. So Library Manager get confused which library you have installed and shows this library as "Updatable". If you "update" it actually installs the marcoschwartz version of the library. However, that is not the cause of the problem you have described in https://github.com/fdebrabander/Arduino-LiquidCrystal-I2C-library/issues/3#issuecomment-210218409. I know the cause of that problem but it is off topic to discuss it here.
>cause they both said fdebrabander

Apparently the marcoschwartz repository is based on this library so they changed the `author` value in library.properties to Frank de Brabander.

tablatronix, 2016-04-15:
Thanks for the explaination i only just started using it, i normally manually install libraries. Looks like i will go back to that.